ROAR-QUTRC/perseus-v2 #28: Can we stop GH Actions building documentation for non-main (to save Action minutes)

DingoOz, 2024-12-20:


RandomSpaceship, 2024-12-20:
We can, but it's intentional (that's how it was originally). Documentation should not be failing depending on the branch it's on. The current build failures are I think specific to the GH Actions runner since I can't replicate them locally, but I managed to fix them by adding a flag to make Electron happy.

RandomSpaceship, 2024-12-21:
Just checked:

> roar-docs-figures> [2503:1220/212225.000583:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /nix/store/b839r5g9ywf4fkama8zah8r55nwbzmky-electron-31.4.0/libexec/electron/chrome-sandbox is owned by root and has mode 4755.

I've fixed this issue in #23, it's not a branch-specific issue, it's Electron having a stroke and will appear on all branches. Given that it's recent, I'm inclined to think that _something_ about the GH Actions runner changed, but adding `--no-sandbox` to the `draw.io` calls fixed it there.

closed 2024-12-21 by RandomSpaceship