ROAR-QUTRC/perseus-v2 #40: Convert `sensors(m2m2)` to use `simple-networking` library

RandomSpaceship, 2024-12-26:
The implementation currently doesn't comply with software standards and will be hard to maintain, as the node in its current state mixes the responsibilities of network I/O and handling the LIDAR's comms protocol.

DingoOz, 2024-12-31:
Will update perseus_sesnors(m2m2_lidar) once advised that simple-networking is merged and stable.

RandomSpaceship, 2024-12-31:
That would have been #39, barring any major issues I don't think it'll be changing significantly from now. The implementation of a `Server` class _might_ change things (in particular the inheritance diagram) but none of the ideas I've got for that right now would change the `Client` usage that you'd need.

DingoOz, 2024-12-31:
Woohoo, thanks

DingoOz, 2025-01-11:
Code using new library has been tested successfully

DingoOz, 2025-01-11:
#72 PR submitted

DingoOz, 2025-01-12:
Changes pushed to address #72 feedback.

closed 2025-01-12 by RandomSpaceship