ROAR-QUTRC/perseus-v2 #42: Investigate Gazebo simulation to support autonomy

DingoOz, 2024-12-26:
Autonomy will require copious amounts of testing, some of which might benefit from simulation.

Gazebo Fortress is the preferred simulation for ROS2 Jazzy, and it is worth investigating the feasibility of implementing it with Perseus.

IssacSim requires nVidia hardware and can be considered in parallel.

RandomSpaceship, 2024-12-26:
I love the idea of running simulations with Gazebo, but I'm not super enthused about Isaac Sim and the like. If it ties us to specific hardware, it just makes it so much harder to test, as well as making it a pain for everyone who might not have that hardware to develop software.

Definitely needs investigation though, maybe Isaac Sim is good enough to justify the trouble.

DingoOz, 2024-12-26:
Yes, I agree focus should be on Gazebo only at first.

Hopefully a Gazebo implementation done smartly doesn't block other simulations laters being integrated

DingoOz, 2025-01-06:
Current blocker: [image]

RandomSpaceship, 2025-01-18:
Simulation successfully demonstrated on branch `gazebo-sim`. Not quite ready for merging yet due to a last couple of minor issues (rover doesn't want to turn), but almost there.

closed 2025-01-18 by RandomSpaceship