ROAR-QUTRC/perseus-v2 #50: Should we be using fast-forward only commits?

RandomSpaceship, 2025-01-03:
That is, should we stop using merge commits?

The pros of FF-only pretty much all come from the fact that each commit only has a single parent:
- Simpler history - easier to read
- Easy to rewrite/cherrypick commits if needed[^1]
- `git diff`s are slightly easier

[^1] Then again, you shouldn't do that on main. In fact, pushing (and force-pushing) to main is explicitly disabled in this repo, so you can't anyway.

The cons are fairly major though:
- Requires people to know how to use Git fairly well (harder for newcomers)
- Branches will need to be rebased every time there's an update on main
- The merge queue may require manual intervention to resolve rebase issues if multiple PRs happen simultaneously (which might not otherwise be a problem with merges)

Personally, I'm against it, and see it as an extra maintenance burden and barrier to entry which really just doesn't need to be there.

DingoOz, 2025-01-03:
I suspect that the cons outweigh the pros. With hopefully more people contributing over time, I think keeping it aligned to standard git practices will mean less admin burden on the existing team. 

Also I hope that as time goes on we can use shorter and shorter lived branches as part of a push to Trunk-based development. I suspect FF-only commits would take us in the opposite direction.

As we start to do Trunk-based development, hopefully our branches should be so short-lived that the benefits of FF-only (like easier git diff) are outweighed by the integration friction.

RandomSpaceship, 2025-01-04:
Yeah agreed. I'll leave this open for a while just in case anybody else wants to weigh in.

RandomSpaceship, 2025-01-09:
Given that we've had no other comments, I'm going to close the issue, and say that we don't really care. The benefits (cleaner tree) don't really outweigh the extra maintenance and learning burden, and it also poses another potential obstacle to newcomers.

closed 2025-01-09 by RandomSpaceship